scottlamb/retina #1: Build Fails

jlpoolen, 2021-06-04:
git cloned my forked copy (jlpooleen/retina) into /usr/local/src

     cd retina
     cargo build
     ...
       Compiling retina v0.0.1 (/usr/local/src/retina)
    error[E0658]: use of unstable library feature 'str_split_once': newly added
       --> src/client/parse.rs:273:18
        |
    273 |                 .split_once(' ')
        |                  ^^^^^^^^^^
        |
        = note: see issue #74773 <https://github.com/rust-lang/rust/issues/74773> for more information

    error[E0658]: use of unstable library feature 'str_split_once': newly added
       --> src/client/parse.rs:285:18
        |
    285 |                 .split_once(' ')
        |                  ^^^^^^^^^^
        |
        = note: see issue #74773 <https://github.com/rust-lang/rust/issues/74773> for more information

    error[E0658]: use of unstable library feature 'str_split_once': newly added
       --> src/client/parse.rs:310:18
        |
    310 |                 .split_once('/')
        |                  ^^^^^^^^^^
        |
        = note: see issue #74773 <https://github.com/rust-lang/rust/issues/74773> for more information

    error[E0658]: use of unstable library feature 'str_split_once': newly added
       --> src/client/parse.rs:553:18
        |
    553 |                 .split_once('=')
        |                  ^^^^^^^^^^
        |
        = note: see issue #74773 <https://github.com/rust-lang/rust/issues/74773> for more information

    error[E0658]: use of unstable library feature 'str_split_once': newly added
       --> src/codec/aac.rs:323:14
        |
    323 |             .split_once('=')
        |              ^^^^^^^^^^
        |
        = note: see issue #74773 <https://github.com/rust-lang/rust/issues/74773> for more information

    error[E0658]: use of unstable library feature 'str_split_once': newly added
       --> src/codec/h264.rs:472:41
        |
    472 |             let (key, value) = p.trim().split_once('=').unwrap();
        |                                         ^^^^^^^^^^
        |
        = note: see issue #74773 <https://github.com/rust-lang/rust/issues/74773> for more information

    error: aborting due to 6 previous errors

    For more information about this error, try `rustc --explain E0658`.
    error: could not compile `retina`

    To learn more, run the command again with --verbose.
    warning: build failed, waiting for other jobs to finish...
    error: build failed
    ares /usr/local/src/retina #

Followed suggested explanation of error:

    ares /usr/local/src/retina # rustc --explain E0658
    #[repr(u128)] // error: use of unstable library feature 'repr128'
    enum Foo {
        Bar(u64),
    }
    ```

    If you're using a stable or a beta version of rustc, you won't be able to use
    any unstable features. In order to do so, please switch to a nightly version of
    rustc (by using [rustup]).

    If you're using a nightly version of rustc, just add the corresponding feature
    to be able to use it:

    ```
    #![feature(repr128)]

    #[repr(u128)] // ok!
    enum Foo {
        Bar(u64),
    }
    ```

    [rustup]: https://rust-lang.github.io/rustup/concepts/channels.html

Here's what I have installed (high watermark from Gentoo repository):

    ares /usr/local/src/retina # eix -I rust
    [I] app-eselect/eselect-rust
         Available versions:  20200419
         Installed versions:  20200419(16:34:41 02/07/21)
         Homepage:            https://wiki.gentoo.org/wiki/Project:Rust
         Description:         Eselect module for management of multiple Rust versions

    [I] dev-lang/rust
         Available versions:  (stable) 1.51.0-r2(stable/1.51)^t ~1.52.1(stable/1.52)^t
           {clippy debug doc miri nightly parallel-compiler rls rustfmt system-bootstrap system-llvm test verify-sig wasm ABI_MIPS="n32 n64 o32" ABI_S390="32 64" ABI_X86="32 64 x32" CPU_FLAGS_X86="sse2" LLVM_TARGETS="AArch64 AMDGPU ARM AVR BPF Hexagon Lanai MSP430 Mips NVPTX PowerPC RISCV Sparc SystemZ WebAssembly X86 XCore"}
         Installed versions:  1.51.0-r2(stable/1.51)^t(22:22:19 05/12/21)(-clippy -debug -doc -miri -nightly -parallel-compiler -rls -rustfmt -system-bootstrap -system-llvm -test -verify-sig -wasm ABI_MIPS="-n32 -n64 -o32" ABI_S390="-32 -64" ABI_X86="64 -32 -x32" CPU_FLAGS_X86="sse2" LLVM_TARGETS="X86 -AArch64 -AMDGPU -ARM -AVR -BPF -Hexagon -Lanai -MSP430 -Mips -NVPTX -PowerPC -RISCV -Sparc -SystemZ -WebAssembly -XCore")
         Homepage:            https://www.rust-lang.org/
         Description:         Systems programming language from Mozilla

    [I] virtual/rust
         Available versions:  1.51.0 ~1.52.1 {ABI_MIPS="n32 n64 o32" ABI_S390="32 64" ABI_X86="32 64 x32"}
         Installed versions:  1.51.0(22:23:52 05/12/21)(ABI_MIPS="-n32 -n64 -o32" ABI_S390="-32 -64" ABI_X86="64 -32 -x32")
         Description:         Virtual for Rust language compiler

    Found 3 matches
    ares /usr/local/src/retina #

    ares /usr/local/src/retina # cargo -V
    cargo 1.51.0
    ares /usr/local/src/retina #

jlpoolen, 2021-06-04:
And, just for completeness, I only have one version staged on my system:

    ares /usr/local/src/retina # eselect rust list
    Available Rust versions:
      [1]   rust-1.51.0 *
    ares /usr/local/src/retina #

scottlamb, 2021-06-04:
The `split_once` it's complaining about was introduced in Rust 1.52 (latest stable version).

jlpoolen, 2021-06-04:
unmasked virtual/rust and dev-lang/rust in Gentoo in /etc/portage/packages.accept_keywords/rust.  Current high watermark for rust in Gentoo is 1.51.[x] :
 
    #
    # 6/4/21 jlpoole: required by retina [Moonfire]
    #
    >=virtual/rust-1.52.1 ~amd64
    >=dev-lang/rust-1.52.1 ~amd64

and emerged virutal/rust which brought in rust 1.52.1.

retina successfully compiled.

Also researched if cargo can specify minimum version for rust and determined it cannot -- such a feature is in the Request For Comments phase:  https://github.com/rust-lang/rfcs/pull/2495

Thanks.

closed 2021-06-04 by jlpoolen