scottlamb/retina #2: Kill Signal For Intended Termination

jlpoolen, 2021-06-05:
I'm creating a Perl script to launch multiple sessions of retina against an array of cameras.  The purpose of this endeavor is to isolate error conditions and create a data set of unique error conditions which may be addressed in your development.

I just sent a "kill -HUP [process #}" and checked my output log of the process so-killed and there was no entry.  What kind of signal should I send to a background running retina process so that it will gracefully shut down and the log will reflect the requested shutdown?   

Here's a current example of the command I'm forking off (from a Perl script using "system($cmd)"):

     cargo run --example client mp4 /tmp/retina/Garage_East_20210605_073201.mp4 --url rtsp://192.168.1.49:554/h264Preview_01_main --username retina --password testingisfun  >/tmp/retina/Garage_East_20210605_073201.log  2>&1 &

I haven't, but will, isolate STD and STDERR, to see if the regular output of each frame can be isolated from error conditions.

Here's the last 5 lines of a log of a process launched in the format above where I sent the active process a "kill -HUP [process number]":

```
jlpoole@ares /usr/local/src/retina $ tail -n 5 /tmp/retina/Peck_West_Alley_20210605_073201.log
2729606549 (mod-2^32: 2729606549), npt 3820.544: 512-byte audio frame
3003918369 (mod-2^32: 3003918369), npt 3820.507: 21397-byte video frame
3003921339 (mod-2^32: 3003921339), npt 3820.540: 22484-byte video frame
3003924309 (mod-2^32: 3003924309), npt 3820.573: 20732-byte video frame
2729607573 (mod-2^32: 2729607573), npt 3820.608: 512-byte audio frame
jlpoole@ares
/usr/local/src/retina $
```

scottlamb, 2021-06-05:
`kill -INT` should work. The CLI is a quick example so I didn't put a lot of thought into different ways people might use it.

jlpoolen, 2021-06-05:
Thank you.  That works.  Here the tail end of my log showing a date time stamp for termination:

```
    Finished dev [unoptimized + debuginfo] target(s) in 0.26s
     Running `target/debug/examples/client mp4 /tmp/retina/Peck_West_Alley_20210605_102220.mp4 --url 'rtsp://192.168.1.52:554/h264Preview_01_main' --username retina --password testingisfun`
I20210605 10:22:45.418 main client::mp4] Stopping
I20210605 10:22:45.420 main client] Done
jlpoole@ares /usr/local/src/retina $

```

closed 2021-06-05 by jlpoolen