scottlamb/retina #13: packet follows marked packet with same timestamp

lucaszanella, 2021-07-27:
`cargo run --example client mp4 --url rtsp://192.168.1.198:10554/tcp/av0_0 --username admin --password 123456 --initial-timestamp ignore /home/dev/orwell/video_samples/tmp/cam_test.mp4`

gives

`E20210727 19:24:02.308 main client] Fatal: [172.17.0.2:50402(me)->192.168.1.198:10554@2021-07-27T19:23:55, 119614@2021-07-27T19:24:02, channel=0, stream=0, ssrc=0000173a] packet follows marked packet with same timestamp`

from here https://github.com/scottlamb/retina/blob/b1db9a9e8b94ff7077050cc26be0a50cbf1bd58e/src/codec/h264.rs#L181

I don't know why `--initial-timestamp ignore ` is needed but without it I get 

https://github.com/scottlamb/retina/blob/b1db9a9e8b94ff7077050cc26be0a50cbf1bd58e/src/client/mod.rs#L705

Anyways, the `packet follows marked packet with same timestamp` error occurs after around 30 calls to `push`, like this:

```
push
push
push
push
push
push
push
push
push
push
push
push
push
push
push
push
8023 (mod-2^32: 8023), npt 0.000: 111304-byte video frame
push
E20210727 19:36:04.662 main client] Fatal: [172.17.0.2:50572(me)->192.168.1.198:10554@2021-07-27T19:36:01, 113756@2021-07-27T19:36:04, channel=0, stream=0, ssrc=00001f57] packet follows marked packet with same timestamp
```

as you see, a video frame is generated and then we get the error. It always happens after the first video frame is emitted.

The strange thing is that on my app, the retina client does not produce this error. I'm trying to figure out what is different.

However, on my app, while the frame producing works, passing `retina::codec::VideoFrame::data().borrow()` to the ffmpeg nal units parser sometimes reject, and sometimes parse and send to the decoder, which procues

```
[h264 @ 0x559374083fc0] non-existing PPS 16 referenced
[h264 @ 0x559374083fc0] Invalid NAL unit 0, skipping.
[h264 @ 0x559374083fc0] Invalid NAL unit 0, skipping.
[h264 @ 0x559374083fc0] Invalid NAL unit 0, skipping.
[h264 @ 0x559374083fc0] Invalid NAL unit 0, skipping.
[h264 @ 0x559374083fc0] no frame!
```

or

```
[h264 @ 0x559374083fc0] non-existing PPS 20 referenced
[h264 @ 0x559374083fc0] Invalid NAL unit 0, skipping.
[h264 @ 0x559374083fc0] no frame!
```

Here's one `VideoFrame`:

`[2021-07-27T19:52:20Z INFO  liborwell::rtsp::retina_client] video frame: VideoFrame { timestamp: 189124 (mod-2^32: 189124), npt 2.061, start_ctx: RtspMessageContext { pos: 42441, received_wall: WallTime(Timespec { sec: 1627415540, nsec: 361449153 }), received: Instant { tv_sec: 421014, tv_nsec: 736674930 } }, end_ctx: RtspMessageContext { pos: 42441, received_wall: WallTime(Timespec { sec: 1627415540, nsec: 361449153 }), received: Instant { tv_sec: 421014, tv_nsec: 736674930 } }, loss: 0, new_parameters: None, is_random_access_point: false, is_disposable: false, data_len: 383 }`

Should I pass the `VideoFrame` `data().borrow()` to ffmpeg like this?

Do you have any idea on the error on the mp4 example, and on my client?

scottlamb, 2021-07-27:
> I don't know why `--initial-timestamp ignore` is needed

That's a consequence of the missing `RTP-Info` header (#10), which contains this timestamp. `permissive` should also work. `require` will fail. `default` will work with a single stream but not more.

> `packet follows marked packet with same timestamp`

Hmm. My understanding from RFC 6184 is that each "access unit" (picture, more or less video frame) should have a different timestamp, and a packet with a MARK indicates that an access unit is complete. (The encoder is allowed to never set a mark, instead just letting the decoder figure that out from the timestamps, but setting MARK means that the decoder knows a frame ended without waiting for the next one to start.) Thus my expectation that the next packet has a different timestamp.

You could try commenting out the if statement that returns this error and see if it helps.

Is it possible the camera uses interlaced video? Then I think there might actually be two pictures per video frame.

Could you post the debug output of a `VideoParameters` from this stream? That might help me figure out what's going on. Actually, the best thing would be a full packet capture if you're willing to share that.

> Should I pass the VideoFrame data().borrow() to ffmpeg like this?

I haven't tried feeding video directly from retina to ffmpeg yet, but in principle it should work. The frames should be fine to pass to ffmpeg. How are you setting up the stream with ffmpeg? You'll likely need to pass it the `extra_data` from `VideoParameters`.

The log messages from ffmpeg suggest it's not seeing a valid stream—NAL unit types should never be 0, and I think it's rare for the PPS id to be 16 rather than 0. But maybe the problem is just that without the extra data, ffmpeg is expecting a stream in Annex B format, and my code is passing it instead in AVC format. (The former means that NAL units are separated by the bytes `00 00 01`, and the latter means that each NAL unit is preceded by its length in bytes as a big-endian number which can be 2, 3, or 4 bytes long. My code uses 4 bytes.) If you prefer to get Annex B data, it'd be possible to add a knob to retina to tell it that. Or conversion isn't terribly difficult: you can scan through NAL units and change the prefix to be `00 00 01`.

I suppose I could add a `retina` example that decodes with ffmpeg into raw images or something. What `ffmpeg` crate are you using?

When you don't get the `packet follows marked packet with same timestamp` error, have you tried saving a `.mp4` and playing it back in your favorite video player? Does it work?

lucaszanella, 2021-07-28:
For ffmpeg I'm using https://github.com/lucaszanella/rust-ffmpeg-1 which uses https://github.com/lucaszanella/rust-ffmpeg-sys-1 (this one is not needed, I just added some vdpau linking stuff, the original could be used). I had to modify the `rust-ffmpeg-1` to add support for ffmpeg's `av_parser_parse2` which parses the individual nal units. The original project doe snot have this and he doesn't want to maintain. My patch is very experimental.

> I haven't tried feeding video directly from retina to ffmpeg yet, but in principle it should work. The frames should be fine to pass to ffmpeg. How are you setting up the stream with ffmpeg? You'll likely need to pass it the extra_data from VideoParameters.

I've never needed to pass additional parameters to ffmpeg, just the nal units. I extracted the h264 bitstream from a big buck bunny .mp4 file and passed to ffmpeg calling `av_parser_parse2` to break into individual nal units and then passed those units using `avcodec_send_packet` and it works. The same process is not working for retina. When my code used to be all C++, I used to pass the output of ZLMediaKit to ffmpeg in this way also and it worked.

Even though `av_parser_parse2` has the option to pass pts, dts, etc, I never used but I'll read more about these parameters. 

`VideoParameters` debug:

```
Some(Video(VideoParameters { rfc6381_codec: "avc1.4D002A", pixel_dimensions: (1920, 1080), pixel_aspect_ratio: None, frame_rate: Some((2, 15)), extra_data: Length: 41 (0x29) bytes
0000:   01 4d 00 2a  ff e1 00 1a  67 4d 00 2a  9d a8 1e 00   .M.*....gM.*....
0010:   89 f9 66 e0  20 20 28 00  00 03 00 08  00 00 03 00   ..f.  (.........
0020:   7c 20 01 00  04 68 ee 3c  80                         | ...h.<. }))
```

I've sent you a dump of the camera via email.

> If you prefer to get Annex B data, it'd be possible to add a knob to retina to tell it that. Or conversion isn't terribly difficult: you can scan through NAL units and change the prefix to be 00 00 01.

do you have experience in which types the rtsp clients out there do these things? I've never took a deep look on how ZLMediaKit does, I simply used it and now I'm getting deeper into RTSP/RTP/h264/etc because rust had no rtsp clients so I had to make one. 

This is how I extracted the big buck bunny to make it work:

`ffmpeg -i BigBuckBunny_512kb.mp4 -vbsf h264_mp4toannexb -vcodec copy -an big_buck_bunny_1280_720.h264`

as you see by `h264_mp4toannexb`, it's as you supposed.

May I know why you use the AVC format in your code? Isn't the  Annex B proper for streaming?

scottlamb, 2021-07-28:
I'm moving the stuff about ffmpeg errors to #15 because I was getting confused mixing them. Leaving this issue as just the `packet follows marked packet with same timestamp` error.

scottlamb, 2021-07-28:
> The strange thing is that on my app, the retina client does not produce this error. I'm trying to figure out what is different.

Is it possible that it's that the `mp4` example sets up both the video and audio stream, while your app only sets up the audio stream? Including the audio stream *shouldn't* cause the video stream to become corrupt, but who knows. You could try passing the `--no-audio` parameter to the `mp4` example and see if that makes any difference.

scottlamb, 2021-07-28:
> Is it possible the camera uses interlaced video? Then I think there might actually be two pictures per video frame.

From the SPS, no. `frame_mbs_only_flag` is 1, so each picture is a frame (rather than a "field").

lucaszanella, 2021-07-30:
It has the same problem if I pass `--no-audio`. In fact I tried this that same day.  

> From the SPS, no. frame_mbs_only_flag is 1, so each picture is a frame (rather than a "field").

it would be good to consider this case for other cameras, thanks

scottlamb, 2021-07-31:
Hmm, I don't know what's different then, unless your app is just getting RTP packets (via `RtspSession`'s `Stream` implementation) rather than H.264 frames (via `RtspSession::demuxed`). You may need to just iteratively make the programs more similar until they get the same behavior and you see what matters, or maybe we can figure it out with a complete packet capture (rather than the ones you got earlier which apparently were missing a lot of IP packets).

>> From the SPS, no. frame_mbs_only_flag is 1, so each picture is a frame (rather than a "field").

> it would be good to consider this case for other cameras, thanks

Yeah, there are certainly various cases on my TODO list to examine, but as far as I can tell none are relevant to your situation.

scottlamb, 2021-08-16:
Did you get any further in understanding this?

I'm also seeing this error on a Reolink RLC-822A camera running firmware v3.0.0.177_21012107. It's just not handling the timestamp and MARK field correctly at all. According to H.264 section 7.4.1.2.3, every access unit must have a VCL NAL, and if there's an SPS and PPS present, they "cannot follow the last VCL NAL unit of the primary coded picture". (In this camera's case, there's one VCL NAL per picture, so the qualifiers "last" and "primary picture" can be ignored.) And [RFC 6184 section 5.1](https://datatracker.ietf.org/doc/html/rfc6184#section-5.1) says that the timestamp must match that of the primary coded picture of the access unit and that the marker bit can only be set on the final packet of the access unit. But the Reolink camera sends at the beginning of the stream:

1. the SPS with the marker bit set (thus putting it into an invalid no-VCL access unit)
2. the PPS (same timestamp, which causes this error)
3. the VCL NAL (SliceLayerWithoutPartitioningIdr) with a new timestamp (so also putting the PPS into an invalid no-VCL access unit)

Then for every subsequent IDR frame, the Reolink camera sends:

1. the SPS with the same timestamp as the *previous* VCL NAL (causing this error again)
2. the PPS, same timestamp
3. the VCL NAL (SliceLayerWithoutPartitioningIdr) with a new timestamp (putting the SPS+PPS into another invalid no-VCL access unit)

Apparently Reolink uses an ancient version of live555. live555's current server code seems to set the marker bit incorrectly (search for `thisNALUnitEndsAccessUnit` in `liveMedia/H264or5VideoStreamFramer.cpp`, which gets set any time the next NAL unit is a SPS or PPS), which probably explains some of the behavior I see.

In a quick look at why other RTSP clients don't seem to care about this:

*   It looks like ffmpeg's `libavformat/rtpdec_h264.c` doesn't handle grouping into access units at all, so it doesn't pay attention to RFC 6184's promises about what NAL units have the marker bit set or have the same timestamp. Instead, I think ffmpeg does the access unit grouping in `libavcodec/h264_parser.c` using just the H.264 section 7.4.1.2.3 rules alone. (Search for `frame_start_found`.) I'm not sure yet, but I think it might it always wait for the next frame to start before emitting the current one, which seems unfortunate.
*   live555 doesn't seem to use the marker bit either on the RTP source side (client side). It might not do any grouping into access units at all, leaving that for the calling application.
*   not sure what vlc (which uses live555) does yet but probably something similar to ffmpeg.

I'm deciding how to handle this. It'd be helpful to know more about when this happens on VStarCam cameras.

lucaszanella, 2021-08-16:
so @scottlamb I'll test more soon on VStarcam, I just need to close https://github.com/scottlamb/retina/pull/12 because I'd like to test and modify things to solve this problem but would be nice to have those patches sent so I can focus on this, otherwise I get confused about different branches and etc. Could you send the .txt for those 2 cases so we both work on the same branch? 

I had some problems these days and I've been doing some things so I have less free time to fix this but I'll certainly investigate this as it's what's missing so I can get images from my camera.

I also need to read more about VCL, PPS, SPS

scottlamb, 2021-08-16:
Ack, I'll look at #12 again tonight.

scottlamb, 2021-08-17:
...tomorrow morning, sorry. These cameras aren't behaving and I'm too sleepy to make sense of it.

commit 6155d4eb (2021-08-19): be more tolerant of incorrect RTP H.264 framing

I'll be optimistic: fixes #13

At least it seems to fix the Reolink case. We'll see about VStarcam.
closed 2021-08-19 by scottlamb