scottlamb/retina #16: Can I read a RTSP stream and publish it to a RTMP server?

seguidor777, 2021-08-04:
First of all thanks a lot for this amazing effort,
I´m looking for a pure Rust RSTP client library for an IoT project and wondering if it´s possible to read a RTSP stream and publish it to a RTMP server. Can you please let me know if this is possible or if it's any plan to support this?

scottlamb, 2021-08-04:
My current goal is to support the rtsp half well, staying agnostic of what happens after that (saving to mp4 as in the current example, decoding with ffmpeg, serving with another protocol, etc). I'm not planning to use rtmp myself right now (although who knows, maybe someone will want it in Moonfire NVR).

 I see there are at least a couple rtmp crates already. I'm very open to any changes to help interoperate with them. For example, as mentioned in #15 we could have a flag to output H.264 video in Annex B format; a rtmp library might be expecting that. I'd also accept a PR with an example that integrates with a rtmp library, assuming it's reasonably small. How does that sound?

scottlamb, 2021-08-04:
I haven't tried any of them, but I see the crate `rtmp` is under recent development which seems promising.

seguidor777, 2021-08-04:
Ok, will take a look. Thanks

seguidor777, 2021-08-04:
Closing as this is not a real issue

closed 2021-08-04 by seguidor777