scottlamb/retina #17: for Reolink cameras: ignore unexpected additional ssrc

scottlamb, 2021-08-14:
As mentioned in [this Apr 29th comment on a moonfire-nvr issue](https://github.com/scottlamb/moonfire-nvr/issues/114#issuecomment-829847828), Reolink cameras have an odd behavior. They sometimes set up a stream with one ssrc, eg:

```
RTP-Info: url=trackID=1;seq=40477;rtptime=2077624148;ssrc=c517011f,url=trackID=2;seq=0;rtptime=0;ssrc=00000000
```

then send RTP packets for both this one *and* another ssrc. I think the latter is for an earlier failed session or something. Sometimes they'll even send the RTP packets for the other ssrc before the connection is even authenticated, which seems like a security hole (if you expose the cameras to an untrusted network, which is probably a bad idea anyway).

Currently when this happens, Retina will fail with an error such as the following (from one of @jlpoolen 's logs today in [this moonfire-nvr issue](https://github.com/scottlamb/moonfire-nvr/issues/144#issue-970985731)):

```
Expected ssrc=Some(21cd90d7) seq=Some(8f4a) got ssrc=6d87df1a seq=88b9 ts=1866533263 (mod-2^32: 1866533263), npt 0.000 at [192.168.1.88:33514(me)->192.168.1.48:554@2021-08-14T11:00:57 pos=1510@2021-08-14T11:00:57]
```

I think we should add an option to so that when we know what `ssrc` to expect, we just (log and) ignore packets with any other `ssrc`. This will likely make these cameras behave acceptably, especially if (as I suspect) the other `ssrc` goes away after a minute or so when the stale session times out.

jlpoolen, 2021-08-15:
A possible tool:  a program that uses the Rust rtsp client that simply opens a connection and tallies the packets received and logs anomalies.  This, along with other items you may have learned about with other cameras, could give a quality report on the camera's ability to keep a stream going (and the libraries ability to handle error/resets).  It could be run outside of moonfire-nvr and left running for a period of time, say up to 24-72 hours.   Almost like a nurse taking vitals before a doctor sees a patient: getting a profile of the camera's ability to send packets conforming to the specifications.

jlpoolen, 2021-08-15:
I was starting to do something like what you looking into in this issue with ffmpeg and the real server when I got interrupted or distracted.  (Having these hardware issues really throws me off track.)  Generating logs has taken some time for me to perfect and I'll go the extra distance (i.e. handling  & preserving ANSI codes) where a lot of prospective users may not.  If there were a simple program that you feed the IP, logon credentials to that would then generate a log on the camera's ability to manage an rtsp feed and its resilience (and that of the network) could help crystallize issues that need consideration or handling and provide a stream-lined path of zeroing in on issues.

scottlamb, 2021-08-19:
This specific issue is in theory fixed in Retina. I'm going to fix another Reolink problem I found on my own test camera then prepare a Moonfire NVR change to pick up both.

Re: tool: did you see [this comment](https://github.com/scottlamb/retina/issues/3#issuecomment-856111889)? It's a little bit like what you're talking about. But I'm not super motivated right now to polish it into something that gets detailed spec compliance info automatically. I have more improvements I want to make than time already, so I'm not going spend a lot of time trying to to get detailed bug reports from people who don't care enough to gather them. I'm prioritizing getting Retina and Moonfire NVR working really well on cameras I have access to (including polished on-NVR and on-camera analytics, live view, and scrub bar UI) over getting it working on every camera out there.

scottlamb, 2021-08-24:
I found a [live555 changelog](http://www.live555.com/liveMedia/public/changelog.txt) entry about this:

```
2015.05.03:
- Updated the "RTSPServer" implementation to fix a bug in RTP/RTCP-over-TCP streaming.
  Before, if the "RTSPClientConnection" object closed before the "RTSPClientSession" object,
  and the TCP connection was also being used for RTP/RTCP-over-TCP streaming, then the
  streaming state (in the "RTSPClientSession") would stay alive, even though the TCP socket
  had closed (and the socket number possibly reused for a subsequent connection).
  This could cause a problem when the "RTSPClientSession" was later reclaimed (due to inactivity).
  Now, whenever a "RTSPClientConnection" object is closed (due to the RTSP TCP connection closing),
  we make sure that we also close any stream that had been using the same TCP connection
  for RTP/RTCP-over-TCP streaming.  (Thanks to Kirill Zhegulev for noting this issue.)
```

The problematic Reolink cameras say `a=tool:LIVE555 Streaming Media v2013.04.08` so no surprise that they're affected by this bug.

My workaround isn't working well: at the timeout of the original session, the camera seems to be sending bytes which aren't not properly framed as RTSP interleaved data, we're dropping the connection, and the cycles repeats.

Another option is to just ensure we send a `TEARDOWN` anywhere possible, and then wait out the session timeout any time it's not possible.

jlpoolen, 2021-08-24:
I'm updating the ip 48 GarageWest camera which is:

     model: RLC-420-5MP
     Hw No.: IPC_51516M5M

and if successfully accomplished, I will test with the Retina client example.

jlpoolen, 2021-08-24:
Blocked.  The PDF included in the zip archive has this:

     Applies to: Reolink RLC-423
     Firmware for hardware version IPC_51516M
with this red warning:

**If your model is not listed above or the hardware version of your camera is not the
exact number listed above, please**

My camera model is RLC-420-5MP and therefore does not match the specified model.

I'll contact Reolink support (and probably be told my camera is "out" of support).

jlpoolen, 2021-08-24:
You have to be careful in selecting which upgrade package.  Firmware number is not sufficient, you also must align to the model number of your camera.  I mistakenly downloaded the first firmware match only read the PDF within the zip archive that the model that upgrade applied to did not match my model.  I then returned to Reolink's site and found several matches to my firmware, and thus discerned which one applied to my camera model.  I applied the patch and rebooted.  Will test this evening with Retina.

scottlamb, 2021-08-24:
Here's the cause of the framing error in the old live555 code:

```c++
Boolean RTPInterface::sendDataOverTCP(int socketNum, u_int8_t const* data, unsigned dataSize, Boolean forceSendToSucceed) {
  if (send(socketNum, (char const*)data, dataSize, 0/*flags*/) != (int)dataSize) {
    // The TCP send() failed.

    if (forceSendToSucceed && envir().getErrno() == EAGAIN) {
      // The OS's TCP send buffer has filled up (because the stream's bitrate has exceeded the capacity of the TCP connection!).
      // Force this data write to succeed, by blocking if necessary until it does:
#ifdef DEBUG_SEND
      fprintf(stderr, "sendDataOverTCP: resending %d-byte send (blocking)\n", dataSize); fflush(stderr);
#endif
      makeSocketBlocking(socketNum);
      Boolean sendSuccess = send(socketNum, (char const*)data, dataSize, 0/*flags*/) == (int)dataSize;
      makeSocketNonBlocking(socketNum);
      return sendSuccess;
    }
    return False;
  }

  return True;
}
```

`send` can partially send the `data`, which happens when the TCP window fills. This code sends the whole thing from the beginning when that happens, which is wrong. This bug has also since been fixed:

```
2013.12.04:
- Updated the "sendDataOverTCP()" function (in "RTPInterface.cpp") to allow for the possibility of
  one of the "send()" calls partially succeeding - i.e., writing some, but not all, of its data.
```

(btw, live555's latest code is still bad. If one client's TCP window fills, the entire server will stall waiting for it to come back.)

scottlamb, 2021-08-24:
I flashed a RLC-420-5MP, hardware version IPC_51516M5M with the [12/11/2020 firmware, aka `v3.0.0.136_20121101`](https://support.reolink.com/hc/en-us/articles/900004823943-12-11-2020-Firmware-for-Reolink-IP-Cameras-IPC-51516M5M-). It still says `a=tool:LIVE555 Streaming Media v2013.04.08`. So this was ancient history for live555 but is a current problem for Reolink cameras.

Here are some options I see:

1.  complain to Reolink support that the latest available firmware for these models has badly flawed support for RTSP/TCP and ask them to update the live555 version. (Unless they have negotiated a special license with live555, they are legally [required](http://www.live555.com/liveMedia/faq.html#copyright-and-license) to comply in countries that care about licenses. But Reolink is Chinese, and I don't think China is such a country...)
2.  switch to RTP/UDP. Normally I'd expect RTP/UDP to be flakier and fussier than RTP/TCP but with these bugs it's likely the reverse. Retina doesn't support RTP/UDP today but it's something I'd like to add eventually.
3.  don't use Reolink's RTSP support at all. Instead use their proprietary protocol (see [Neolink](https://github.com/thirtythreeforty/neolink)) or RTMP. Via an RTSP bridge, or in Moonfire's case I could eventually add direct support for these.
4.  avoid the additional ssrc via teardown or waiting (rather than dropping the extra packets as I tried in d8aa4c6). Hope that without the redundant data, the framing error doesn't come up enough to be a big problem.
5.  make an ugly, old-live555-specific hack to avoid the framing error also: before interpreting an interleaved data packet, wait for the following message. If the second message doesn't parse, figure out how many redundant bytes were added mid-packet by the `sendDataOverTCP()` bug before parsing the first message.

scottlamb, 2021-08-24:
I just wrote to Reolink.

jlpoolen, 2021-08-24:
or 
6. Declare Reolink cameras unsupported at this time.

I'd be interested to learn what other manufacturers use as servers and their conformance to standards.  It seems having Scott spend time on one brand which has been documented to be using.  

I've been using Reolink because they're cheap and if they get stolen or vandalized, it's not going to be expensive to replace.  If other brands provide a lot less problems conforming to standards, then it would be appropriate to weigh their cost.  If Reolinks cost $40, and Brand X costs $70 and works well within the RTSP protocol, then it may make sense to switch brands.  Now, if the next least expensive brand that plays nicely costs $150, then it might merit further consideration.

I think further time Scott spends on Reolink's issues is time away from other enhancements that affect all users of moonfire-nvr.

scottlamb, 2021-08-24:
I'm currently not sure what manufacturers to recommend, though. :-(

*   Dahua, Hikvision, and Uniview are supporting genocide, so they're out regardless of technical merit or value.
*   I have a Geovision camera being shipped to me that will arrive Thursday; I'll try it out and see if it's any better.
*   Hanwha seems a bit pricy; Axis more so.

jlpoolen, 2021-08-25:
I flashed my older one as described above with the correct firmware version for the model I had. I, too, have the session (using the LIVE555 openRTSPclient) showing:

	v=0
	o=- 1629795645777963 1 IN IP4 192.168.1.48
	s=Session streamed by "preview"
	i=h264Preview_01_main
	t=0 0
	a=tool:LIVE555 Streaming Media v2013.04.08
	a=type:broadcast
	a=control:*
	a=range:npt=0-
	a=x-qt-text-nam:Session streamed by "preview"
	a=x-qt-text-inf:h264Preview_01_main
	m=video 0 RTP/AVP 96
	c=IN IP4 0.0.0.0
	b=AS:500
	a=rtpmap:96 H264/90000
	a=fmtp:96 packetization-mode=1;profile-level-id=640033;sprop-parameter-sets=Z2QAM6zoAoALWQ==,aO48MA==
	a=control:trackID=1
	m=audio 0 RTP/AVP 97
	c=IN IP4 0.0.0.0
	b=AS:256
	a=rtpmap:97 MPEG4-GENERIC/16000
	a=fmtp:97 streamtype=5;profile-level-id=15;mode=AAC-hbr;sizelength=13;indexlength=3;indexdeltalength=3;config=1408; profile=1;
	a=control:trackID=2

	Opened URL "rtsp://192.168.1.48:554/h264Preview_01_main", returning a SDP description:
	v=0
	o=- 1629795645777963 1 IN IP4 192.168.1.48
	s=Session streamed by "preview"
	i=h264Preview_01_main
	t=0 0
	a=tool:LIVE555 Streaming Media v2013.04.08
	a=type:broadcast
	a=control:*
	a=range:npt=0-
	a=x-qt-text-nam:Session streamed by "preview"
	a=x-qt-text-inf:h264Preview_01_main
	m=video 0 RTP/AVP 96
	c=IN IP4 0.0.0.0
	b=AS:500
	a=rtpmap:96 H264/90000
	a=fmtp:96 packetization-mode=1;profile-level-id=640033;sprop-parameter-sets=Z2QAM6zoAoALWQ==,aO48MA==
	a=control:trackID=1
	m=audio 0 RTP/AVP 97
	c=IN IP4 0.0.0.0
	b=AS:256
	a=rtpmap:97 MPEG4-GENERIC/16000
	a=fmtp:97 streamtype=5;profile-level-id=15;mode=AAC-hbr;sizelength=13;indexlength=3;indexdeltalength=3;config=1408; profile=1;
	a=control:trackID=2

I tried several sessions using the Retina client and continued to have problems.  The best session would be about 16 seconds and then I'd get an error message.  Examples:

	282699224 (mod-2^32: 282699224), npt 11.968: 249-byte audio frame
	652994919 (mod-2^32: 652994919), npt 11.900: 29782-byte video frame
	282700248 (mod-2^32: 282700248), npt 12.032: 249-byte audio frame
	652997889 (mod-2^32: 652997889), npt 11.933: 30061-byte video frame
	653000859 (mod-2^32: 653000859), npt 11.966: 30276-byte video frame
	E20210824 17:30:42.477 main client] Fatal: [192.168.1.28:48462(me)->192.168.1.48:554@2021-08-24T17:30:29, 12688689@2021-08-24T17:30:42] RTSP framing error: Invalid RTSP message; next 128 of 790 buffered bytes are:
	0000:   0b 32 1a 41  68 15 23 ea  79 50 6c e3  d1 f7 6f a2   .2.Ah.#.yPl...o.
	0010:   6e 61 82 e0  2a db df 8c  d4 7a 3d 11  fa 38 3e c5   na..*....z=..8>.
	0020:   9d 02 84 cb  49 5e c4 d4  bb b0 25 16  57 50 24 b6   ....I^....%.WP$.
	0030:   76 98 91 26  60 90 ff a2  d5 01 da c2  ce d2 56 5f   v..&`.........V_
	0040:   79 39 a5 02  b5 ae 9c 92  44 3e 42 2d  72 28 6d 84   y9......D>B-r(m.
	0050:   2b a1 8e 5d  cc 8d 89 23  29 05 f7 7a  4e d8 48 5b   +..]...#)..zN.H[
	0060:   ed d3 07 43  a1 7f 25 9b  b4 6f e1 90  4d 96 37 61   ...C..%..o..M.7a
	0070:   45 f7 1d 6a  ad d4 4c cc  3b c2 45 35  21 c8 27 f3   E..j..L.;.E5!.'.
	I20210824 17:30:42.478 main client] Done
	Tue Aug 24 05:30:42 PM PDT 2021
	jlpoole@taurus /usr/local/src/retina $ 
[... 153 lines cut ...]
	Received 251 new bytes of response data.
	Received a complete PLAY response:
	RTSP/1.0 200 OK
	Server: Rtsp Server/2.0
	CSeq: 7
	Date: Wed, Aug 25 2021 02:21:10 GMT
	Range: npt=0.000-
	Session: 54C4E3AC
	RTP-Info: url=trackID=1;seq=8202;rtptime=718423480;ssrc=d89a16ec,url=trackID=2;seq=51872;rtptime=3942782428;ssrc=6aca52b4

	Started playing session
	Receiving streamed data (for up to 600.000000 seconds)...
	Sending request: TEARDOWN rtsp://192.168.1.48/h264Preview_01_main/ RTSP/1.0
	CSeq: 8
	Authorization: Digest username="moon", realm="LIVE555 Streaming Media", nonce="435a79a4baaee7fe09baf6278cfe05ce", uri="rtsp://192.168.1.48/h264Preview_01_main/", response="00429a724edcc215d201530ed4392596"
	User-Agent: ./openRTSP (LIVE555 Streaming Media v2021.08.24)
	Session: 54C4E3AC

	Received 65 new bytes of response data.
	Received a complete TEARDOWN response:
	RTSP/1.0 200 OK
	CSeq: 8
	Date: Wed, Aug 25 2021 02:31:10 GMT

	jlpoole@taurus /usr/local/src/Live/live/testProgs $

scottlamb, 2021-09-16:
For the Reolink RLC-420-5MP / hardware version IPC_51516M5M: at my request, Reolink built [firmware v3.0.0.589_21091583](https://reolink.zendesk.com/attachments/token/HkpMqwJfjdnSOxAmn5pm6w9rZ/?name=IPC_51516M5M.589_21091583.RLC-420-5MP.SC5035.5MP.REOLINK.pak) which advertises `a=tool:LIVE555 Streaming Media v2020.08.12`. It no longer has the stale TCP session bug, so there's no need to wait 65 seconds after an unclean shutdown to connect again.

They tell me that the Reolink RLC-410 / hardware version IPC_3816M is too old and has been discontinued, so they won't be updating its firmware.

jlpoolen, 2021-09-16:
For posterity, I purchased directly from Reolink 2 RLC-410 5mp cameras on 7/14/2021; $76.98 inc. shipping for 2 cameras.  They arrived 7/24/21.  

Here is the firmware they currently have
[image]
:

jlpoolen, 2021-09-17:
I ran ffmpeg (use "-loglevel debug" mode to get the "a=tool:" at line 45)  and got:

	     1  Commenced Thu Sep 16 08:48:17 PM PDT 2021
	     2  Command:
	     3     ffmpeg -loglevel debug -fflags nobuffer -i rtsp://moon:fire123@192.168.1.49:554/h264Preview_01_main -t 15 -y /tmp/Sep_16_204817.mp4 
	     4     
            ...
   
	    38  [tcp @ 0x55dec43e9640] Successfully connected to 192.168.1.49 port 554
	    39  [rtsp @ 0x55dec43e6700] SDP:
	    40  v=0
	    41  o=- 1631437242087047 1 IN IP4 192.168.1.49
	    42  s=Session streamed by "preview"
	    43  i=h264Preview_01_main
	    44  t=0 0
	    45  a=tool:LIVE555 Streaming Media v2013.04.08
	    46  a=type:broadcast
	    47  a=control:*
	    48  a=range:npt=0-
	    49  a=x-qt-text-nam:Session streamed by "preview"
	    50  a=x-qt-text-inf:h264Preview_01_main
	    51  m=video 0 RTP/AVP 96
	    52  c=IN IP4 0.0.0.0
	    53  b=AS:500
	    54  a=rtpmap:96 H264/90000
	    55  a=range:npt=0-
	    56  a=fmtp:96 packetization-mode=1;profile-level-id=640033;sprop-parameter-sets=Z2QAM6zoAoALWQ==,aO48MA==
	    57  a=control:trackID=1
	    58  m=audio 0 RTP/AVP 97
	    59  c=IN IP4 0.0.0.0
	    60  b=AS:256
	    61  a=rtpmap:97 MPEG4-GENERIC/16000
	    62  a=fmtp:97 streamtype=5;profile-level-id=15;mode=AAC-hbr;sizelength=13;indexlength=3;indexdeltalength=3;config=1408; profile=1;
	    63  a=control:trackID=2
	    64
	    65  Failed to parse interval end specification ''
	    66  [rtsp @ 0x55dec43e6700] video codec set to: h264

It appears that since  July 2021, Reolink has changed its LIVE555 from a 2013 version to a 2020 version.

jlpoolen, 2021-09-17:
I ran ffmpeg to see what the rtsp server version is.  Note, you need to include "-loglevel debug " to get line 45 "a=tool:LIVE555 Streaming Media v2013.04.08"

    jlpoole@taurus /usr/local/src $ ./ffmpeg1.sh 
    Thu Sep 16 08:59:25 PM PDT 2021
    Here is the log: /tmp/stderr_Sep_16_205901.log
     1  Commenced Thu Sep 16 08:59:01 PM PDT 2021
     2  Command:
     3     ffmpeg -loglevel debug -fflags nobuffer -i rtsp://moon:fire123@192.168.1.56:554/h264Preview_01_main -t 15 -y /tmp/Sep_16_205901.mp4 
     4     
     5  ffmpeg version 4.4 Copyright (c) 2000-2021 the FFmpeg developers
     6    built with gcc 10.3.0 (Gentoo 10.3.0-r2 p3)
     ...
    39  [rtsp @ 0x55692ec55700] SDP:
    40  v=0
    41  o=- 1631548228552005 1 IN IP4 192.168.1.56
    42  s=Session streamed by "preview"
    43  i=h264Preview_01_main
    44  t=0 0
    45  a=tool:LIVE555 Streaming Media v2013.04.08
    46  a=type:broadcast
    47  a=control:*
    48  a=range:npt=0-
    49  a=x-qt-text-nam:Session streamed by "preview"
    50  a=x-qt-text-inf:h264Preview_01_main
    51  m=video 0 RTP/AVP 96
    52  c=IN IP4 0.0.0.0
    53  b=AS:500
    54  a=rtpmap:96 H264/90000
    55  a=fmtp:96 packetization-mode=1;profile-level-id=640033;sprop-parameter-sets=Z2QAM6zoAoAPGQ==,aO48sA==
    56  a=control:trackID=1
    57  m=audio 0 RTP/AVP 97
    58  c=IN IP4 0.0.0.0
    59  b=AS:256
    60  a=rtpmap:97 MPEG4-GENERIC/16000
    61  a=fmtp:97 streamtype=5;profile-level-id=15;mode=AAC-hbr;sizelength=13;indexlength=3;indexdeltalength=3;config=1408; profile=1;
    62  a=control:trackID=2
    63
    64  Failed to parse interval end specification ''
    65  [rtsp @ 0x55692ec55700] video codec set to: h264
    66  [rtsp @ 0x55692ec55700] RTP Packetization Mode: 1

**Conclusion**:  The change from 2013 to 2020 of the LIVE555 server in the Reolink camera occurred after July 2021.

scottlamb, 2021-11-30:
I think I'm going to call this done. The major improvements here we've made to date are:
*   supporting UDP, which is less buggy on these models.
*   logging a warning when connecting to these models with TCP
*   trying to send a `TEARDOWN` when ending even a TCP session to these cameras, to avoid triggering the stale session bug
*   recognizing stale sessions (either by failed `TEARDOWN` or detected by the erroneous packets) and tracking their expiration. Moonfire NVR explicitly waits them out; other callers can do the same.

I haven't added any kind of "resync" to recover from these cameras' buggy behavior after the TCP window fills. Maybe I'll add that yet, but I'm not seeing complaints right now anyway.

closed 2021-11-30 by scottlamb

hn, 2022-03-10:
I would like to kindly point out my findings on Reolink cameras if this is not already known: [Reolink RLC-410-5MP IP camera firmware unpacker and reverse engineered technical details](https://github.com/hn/reolink-camera). Maybe someone is able to patch the camera Linux system so that the Live555 server is updated (... or even replaced with a completely different streaming-server).

jlpoolen, 2022-03-10:
Wow!  I gave this a quick 2 minute scan and felt like I had entered
Heaven.  This is fantastic.  I had contacted George Hillard and he did not
reply so I gave up.  But what you have accomplished here is what I have
dreamt about if i had the time.  I've been pursuing a Raspberyy Pi based
approach using vetted hadware components and then hoping to use gstreamer
and then whatever Scott comes up with that gives a no problem stream, or if
there are problems, they can be identified as to what the cause is, i.e.
long CATV lines causing voltage drops that is afflicting the hardware, or
packet congestion at a node such as the PoE router.

I cannot wait to sink my teeth into your project, though I'm going to help
the GenPI64 project if I can to create Gentoo images for Raspberry Pi so I
have a completely controlled playing field.

As for LIVE555 -- I really do not think the maintainter's approach of
limited transparency is good and possibly a red flag.  I suspect he may
have conflicted interests in having his code used by camera manufacturers
such as Reolink and projects like VLC and therefore is unwilling to stage
the codebase in a repository that makes the history of changes fully
transparent.  That plus his requirement that I must use an email address
other than gmail to post to his list despite his knowing I've been a
developer at a major software company for over 22 years.

I can't wait to carefully read your project.  (I've been embarking on the
side to learn electronics... again so I can look at something like the
Reolink circuit board and have a better understanding of their design.  I
also confess I'm tired of my Zylex switches dying because of a cheap
capacitor in the power board -- twice I hunted down the blown capacitors
and replace them so I could resuce the switch.  I have one now where my
capacitor tests indicate all of the capacitors are okay, so I need to learn
more to determine the power unit has failed and how to fix, or better yet,
replace it.  All this inspired by a posting on the Amazon comments for the
Zylex showing a way on how to replace the blown capacitors.)  I've
digressed, but I do that when i get excited after finding projects such as
yours and Scott's.  I'll be in touch.

John

1566 Court ST NE
Salem, OR 97301-4241
(707) 812-1323

commit d8aa4c65 (2021-08-19): workaround for Reolink spurious data

Fixes #17

I'm accumulating unit testing debt, but I think I'd rather get this
quickly out in the field to test against scenarios I haven't anticipated
than get my unit tests in order for scenarios I have anticiapted.