scottlamb/retina #21: bad access unit

lucaszanella, 2021-08-21:
With the new modifications, I'm getting a stream with no errors but this warning:

```
[2021-08-21T05:32:37Z DEBUG retina::codec::h264] bad access unit (ended by mark) at ts 1147 (mod-2^32: 1147), npt 0.000
    errors are:
    * same timestamp as previous access unit
    NALs are:
      0: NalHeader { nal_ref_idc: 3, nal_unit_type: SliceLayerWithoutPartitioningNonIdr }
```

should it happen? I had no time to look into it deep and I can provide more info if needed. I was just testing if everything went well with the merge. 

I don't get decoded frames yet but I think this has to do with what we talked before about the VCL thing etc. By the way, do you have a place for me to read about those without getting into tedious details about h264 encoding internals?

I cleaned my ffmpeg code, it compiles on the retina branch but have to send packets from retina to it. Gonna try to make the example soon so you can test with your cameras too, if you want

scottlamb, 2021-08-21:
This is the new version of `packet follows marked packet with same timestamp` from #13—now a debug message rather than a hard error, in the interest of improving compatibility. Not sure if the bad data will be good enough for ffmpeg (or whatever). You can see if the `.mp4` generated by the example plays or not.

If you set the log level on `retina::codec::h264` to trace, you'll see log entries for all the access units, not just the ones with errors. It'd be interesting to see the one immediately before. The Reolink camera I'm working with will incorrectly end access units at SPS and PPS NALs, and I added code to prevent that:

https://github.com/scottlamb/retina/blob/59e513c9be90afa52c907839fa0f6c5ceb7fe61c/src/codec/h264.rs#L499-L511

Possibly we should do something similar for the VStarcam.

scottlamb, 2026-10-17:
I believe this has been fixed since #15. If anyone discovers otherwise, let's open a fresh issue.

closed 2026-10-17 by scottlamb