scottlamb/retina #25: support basic authentication

scottlamb, 2021-08-23:
From <https://github.com/scottlamb/moonfire-nvr/discussions/151>. When configured with username and password, Anpviz IPC-D250 reports:

```
W20210823 05:40:40.871 s-ChuckRear-sub moonfire_nvr::streamer] ChuckRear-sub: sleeping for Duration { secs: 1, nanos: 0 } after error: [192.168.254.254:39730(me)->192.168.254.5:554@2021-08-23T05:40:40, 0@2021-08-23T05:40:40] Unauthorized response to DESCRIBE CSeq=1: Non-digest authentication requested: Basic realm="/"
```

scottlamb, 2021-08-23:
Hmm, servers are allowed to specify more than one challenge as a comma-separated list (or as multiple headers, which `rtsp-types` converts to the comma-separated form [here](https://github.com/sdroege/rtsp-types/blob/53bdf9a74175946572eb3509a6343696724123eb/src/headers/types.rs#L83-L84)). Ideally we would support that and pick out digest if it's present, then basic if it's present, then err out otherwise. 

Splitting the `WWW-Authenticate` header into challenges is more than `header_value.split(", ")` because `, ` can occur in a quoted string. See the grammar from [RFC 7235 Appendix C](https://datatracker.ietf.org/doc/html/rfc7235#appendix-C):

```
   WWW-Authenticate = *( "," OWS ) challenge *( OWS "," [ OWS challenge
    ] )

   auth-param = token BWS "=" BWS ( token / quoted-string )
   auth-scheme = token

   challenge = auth-scheme [ 1*SP ( token68 / [ ( "," / auth-param ) *(
    OWS "," [ OWS auth-param ] ) ] ) ]
   credentials = auth-scheme [ 1*SP ( token68 / [ ( "," / auth-param )
    *( OWS "," [ OWS auth-param ] ) ] ) ]

   quoted-string = <quoted-string, see [RFC7230], Section 3.2.6>

   token = <token, see [RFC7230], Section 3.2.6>
   token68 = 1*( ALPHA / DIGIT / "-" / "." / "_" / "~" / "+" / "/" )
    *"="
```

We could parse to split it properly, then give the digest part to `digest_auth` to parse again, but that's annoying to parse twice.

@lucaszanella you were working to polish up `www-authenticate` (KeenS/www-authenticate#4). Do you want to finish that? It'd parse everything in one go.

lucaszanella, 2021-08-23:
Yes I can finish it but we'd have to do some reviews and test it with all the cameras. Wouldn't it be easier to just patch the current one? 

My patches to www-authenticate are nice, removed unsafe calls and did some more testing, but it's your call: https://github.com/lucaszanella/www-authenticate/blob/master/src/lib.rs

scottlamb, 2021-08-23:
I'm assuming from the name `digest_auth` that its author isn't interested in expanding it to parse other challenge types like basic.

lucaszanella, 2021-08-23:
No basic? Mine has support for. So what's missing is multiple authentication methods? Can you give me the one that the camera sends?

lucaszanella, 2021-08-27:
I'm confused. Shouldn't these 2 `WWW-Authenticate`  be treated separately and sent to www-authenticate?

Also,

```
    fn test_parse_crazy_answer() {
        let b = b"Basic realm=\"/\"";
        let stream = Stream::new(b);
        match stream.challenge().unwrap() {
            (_, RawChallenge::Token68(_)) => panic!(),
            (scheme, RawChallenge::Fields(fields)) => {
                assert_eq!(scheme, "Basic");
            }
        }
        assert!(stream.is_end());
    }
```

works on my crate

scottlamb, 2021-08-27:
> I'm confused. Shouldn't these 2 `WWW-Authenticate` be treated separately and sent to www-authenticate?

[RFC 2636 section 4.2](https://datatracker.ietf.org/doc/html/rfc2326#section-4.2) refers to [RFC 2616 section 4.2](https://datatracker.ietf.org/doc/html/rfc2616#section-4.2), which says the following:

```
   Multiple message-header fields with the same field-name MAY be
   present in a message if and only if the entire field-value for that
   header field is defined as a comma-separated list [i.e., #(values)].
   It MUST be possible to combine the multiple header fields into one
   "field-name: field-value" pair, without changing the semantics of the
   message, by appending each subsequent field-value to the first, each
   separated by a comma. The order in which header fields with the same
   field-name are received is therefore significant to the
   interpretation of the combined field value, and thus a proxy MUST NOT
   change the order of these field values when a message is forwarded.
```

and `rtsp-types` combines it that way, with the code I linked above.