scottlamb/retina #59: access raw RTCP packet (headers and all)

RustPanda, 2022-05-05:
Hello. To integrate `Gstreame`r and `WebRTC` into `Retina`, need to transfer the raw RTCP packet. Can you add `SenderReport::raw(&self) -> &[u8]`?
```rust
let launch = "rtpsession name=session
    appsrc name=rtpsrc ! session.recv_rtp_sink \
    appsrc name=rtcpsrc ! session.recv_rtcp_sink \  
    session.recv_rtp_src ! rtph264depay ! h264parse ! vaapidecodebin ! autovideosink";
```

scottlamb, 2022-05-05:
I can add something. RTCP comes in "compound packets" which usually start with a sender or receiver report (RFC 3550 says they MUST; RFC 5506 weakens this) and usually end with a source description (similarly, RFC 3550 says so, RFC 5506 says not necessarily). What does gstreamer expect—to get just sender report packets? to get compound packets? to get each packet within the compound packet individually?

RustPanda, 2022-05-05:
@scottlamb

-compound packets can be sent to Gstreamer [GstRTCPBuffer](https://gstreamer.freedesktop.org/documentation/rtplib/gstrtcpbuffer.html?gi-language=c)
- I don't know if  sender report packets  will be enough for him [rtspsession](https://gstreamer.freedesktop.org/documentation/rtpmanager/rtpsession.html?gi-language=c)

scottlamb, 2022-05-10:
I'm working on an updated interface to access the full compound packet.

scottlamb, 2022-05-10:
Now done, on the `next` branch. Please let me know how the new API works out for you.

RustPanda, 2022-05-11:
@scottlamb thanks, I'll check this week

commit 27c5ed5f (2022-05-10): expose RTCP compound packets

...and rename `PacketItem` variants to `Rtp` and `Rtcp` for brevity.

Fixes #59
closed 2022-05-17 by scottlamb