scottlamb/retina #62: client failed to authenticate with Dahua Camera

wangjia184, 2022-06-26:
I have a Dahua camera serving RTSP at `rtsp://192.168.8.20:554/cam/realmonitor?channel=1&subtype=0` 

I tried with correct username and password to connect to, it fails with authentication error.

```
$ cargo run --example client mp4 --url 'rtsp://192.168.8.20:554/cam/realmonitor?channel=1&subtype=0' --username admin --password 123456 out.mp4

E20220626 21:00:27.992 main client] Fatal: Unauthorized response to DESCRIBE CSeq=2: Received Unauthorized after trying digest auth

conn: 192.168.8.10:61108(me)->192.168.8.20:554@2022-06-26T21:00:27
msg: 139@2022-06-26T21:00:27
```

**The credentails are 100% correct.**  I tried to sniffer the communication between retina client and camera, and here is what I captured.

```
DESCRIBE rtsp://192.168.8.20:554/cam/realmonitor?channel=1&subtype=0 RTSP/1.0\r\n
Accept: application/sdp\r\n
CSeq: 1\r\n
User-Agent: Retina mp4 example\r\n
\r\n

RTSP/1.0 401 Unauthorized\r\n
CSeq: 1\r\n
WWW-Authenticate: Digest realm="Login to 4K01FD2PAJ6E3FF", nonce="203c8d27b595204504dfa65fd22586bd"\r\n
\r\n

DESCRIBE rtsp://192.168.8.20:554/cam/realmonitor?channel=1&subtype=0 RTSP/1.0\r\n
Accept: application/sdp\r\n
Authorization: Digest username="admin", realm="Login to 4K01FD2PAJ6E3FF", uri="rtsp://192.168.8.20:554/cam/realmonitor?channel=1&subtype=0", nonce="203c8d27b595204504dfa65fd22586bd", response="61dd3463aff71f24ee30ca23e3eeae4a"\r\n
CSeq: 2\r\n
User-Agent: Retina mp4 example\r\n
\r\n

RTSP/1.0 401 Unauthorized\r\n
CSeq: 2\r\n
WWW-Authenticate: Digest realm="Login to 4K01FD2PAJ6E3FF", nonce="203c8d27b595204504dfa65fd22586bd"\r\n
\r\n

```
----------

While the same RTSP url and credentials work fine in VLC.  I also sniffered the communication between VLC and camera, the following are the captured packets.

```
OPTIONS rtsp://192.168.8.20:554/cam/realmonitor?channel=1&subtype=0 RTSP/1.0\r\n
CSeq: 2\r\n
User-Agent: LibVLC/3.0.17.3 (LIVE555 Streaming Media v2016.11.28)\r\n
\r\n

Response: RTSP/1.0 401 Unauthorized\r\n
CSeq: 2\r\n
WWW-Authenticate: Digest realm="Login to 4K01FD2PAJ6E3FF", nonce="e8ca006dd506cf4fdd75de1950f5aeab"\r\n
\r\n

OPTIONS rtsp://192.168.8.20:554/cam/realmonitor?channel=1&subtype=0 RTSP/1.0\r\n
CSeq: 3\r\n
Authorization: Digest username="admin", realm="Login to 4K01FD2PAJ6E3FF", nonce="e8ca006dd506cf4fdd75de1950f5aeab", uri="rtsp://192.168.8.20:554/cam/realmonitor?channel=1&subtype=0", response="f8d0870b5bf8db1967c1a04587c04fb2"\r\n
User-Agent: LibVLC/3.0.17.3 (LIVE555 Streaming Media v2016.11.28)\r\n
\r\n

RTSP/1.0 401 Unauthorized\r\n
CSeq: 3\r\n
WWW-Authenticate: Digest realm="Login to 4K01FD2PAJ6E3FF", nonce="e8ca006dd506cf4fdd75de1950f5aeab"\r\n
\r\n

DESCRIBE rtsp://192.168.8.20:554/cam/realmonitor?channel=1&subtype=0 RTSP/1.0\r\n
CSeq: 4\r\n
Authorization: Digest username="admin", realm="Login to 4K01FD2PAJ6E3FF", nonce="e8ca006dd506cf4fdd75de1950f5aeab", uri="rtsp://192.168.8.20:554/cam/realmonitor?channel=1&subtype=0", response="f7d5cde2d331183cb1710e50d9738971"\r\n
User-Agent: LibVLC/3.0.17.3 (LIVE555 Streaming Media v2016.11.28)\r\n
Accept: application/sdp\r\n
\r\n

RTSP/1.0 401 Unauthorized\r\n
CSeq: 4\r\n
WWW-Authenticate: Digest realm="Login to 4K01FD2PAJ6E3FF", nonce="e8ca006dd506cf4fdd75de1950f5aeab"\r\n
\r\n

Request: OPTIONS rtsp://192.168.8.20:554/cam/realmonitor?channel=1&subtype=0 RTSP/1.0\r\n
CSeq: 5\r\n
User-Agent: LibVLC/3.0.17.3 (LIVE555 Streaming Media v2016.11.28)\r\n
\r\n

RTSP/1.0 401 Unauthorized\r\n
CSeq: 5\r\n
WWW-Authenticate: Digest realm="Login to 4K01FD2PAJ6E3FF", nonce="e8ca006dd506cf4fdd75de1950f5aeab"\r\n
\r\n

OPTIONS rtsp://192.168.8.20:554/cam/realmonitor?channel=1&subtype=0 RTSP/1.0\r\n
CSeq: 6\r\n
Authorization: Digest username="admin", realm="Login to 4K01FD2PAJ6E3FF", nonce="e8ca006dd506cf4fdd75de1950f5aeab", uri="rtsp://192.168.8.20:554/cam/realmonitor?channel=1&subtype=0", response="ad87c66be128206abbd2dd51fea8e563"\r\n
User-Agent: LibVLC/3.0.17.3 (LIVE555 Streaming Media v2016.11.28)\r\n
\r\n

RTSP/1.0 200 OK\r\n
CSeq: 6\r\n
Server: Rtsp Server/3.0\r\n
Public: OPTIONS, DESCRIBE, ANNOUNCE, SETUP, PLAY, RECORD, PAUSE, TEARDOWN, SET_PARAMETER, GET_PARAMETER\r\n
\r\n

DESCRIBE rtsp://192.168.8.20:554/cam/realmonitor?channel=1&subtype=0 RTSP/1.0\r\n
CSeq: 7\r\n
Authorization: Digest username="admin", realm="Login to 4K01FD2PAJ6E3FF", nonce="e8ca006dd506cf4fdd75de1950f5aeab", uri="rtsp://192.168.8.20:554/cam/realmonitor?channel=1&subtype=0", response="f6cbb524ce35db485250b8b88445365c"\r\n
User-Agent: LibVLC/3.0.17.3 (LIVE555 Streaming Media v2016.11.28)\r\n
Accept: application/sdp\r\n
\r\n

RTSP/1.0 200 OK\r\n
CSeq: 7\r\n
x-Accept-Dynamic-Rate: 1\r\n
Content-Base: rtsp://192.168.8.20:554/cam/realmonitor?channel=1&subtype=0/\r\n
Cache-Control: must-revalidate\r\n
Content-length: 630
Content-type: application/sdp
\r\n

```

Bug in digest authentication?

scottlamb, 2022-06-28:
Interesting, thanks for the detailed report.

What model & firmware version is this? It must not be all Dahua cameras; I have a couple that are working.

> it seems VLS is trying an alternative approach to compute digest and then succeeded.

`MD5(HA1:nonce:HA2)` is the older RFC 2069 style. The `http-auth` crate which `retina` uses tries this form of calculation if there's no `qop` specified in the 401 response's `WWW-Authenticate` header. Interesting if VLC tries both.

https://github.com/scottlamb/http-auth/blob/5d8be8808cf6009ea46b10c1eb113d98f16d3361/src/digest.rs#L306

https://github.com/scottlamb/http-auth/blob/5d8be8808cf6009ea46b10c1eb113d98f16d3361/src/digest.rs#L418

scottlamb, 2022-06-28:
Hmm, but it can't be computing in the newer style, because it's not passing along the other necessary parameters like `nc` and `cnonce`. Huh. I don't know what other calculation they'd be doing, and (in a quick skim of vlc and live555 code) I didn't see the code that would do it. Puzzling.

wangjia184, 2022-06-28:
@scottlamb It is a sub-brand of Dahua, IMOU indoor camera TF5

Dont think it is `MD5-sess` as I dont see `cnonce` in the request.  I tried to compute hash manually and cannot get the result `f6cbb524ce35db485250b8b88445365c`.  Asking this question on SO:  https://stackoverflow.com/questions/72774182/rtsp-digest-authentication-in-vlc

scottlamb, 2022-07-15:
Did you get this figured out? I just clicked the SO link and got "This question was voluntarily removed by its author."

wangjia184, 2022-07-16:
yep, I found the reason,  VLC  was trying to use some password from system for retry. so my password was wrong.
closing this issue

closed 2022-07-16 by wangjia184