scottlamb/retina #64: Fatal: Timestamp jumped

christopher-wilke, 2022-08-10:
I am trying to run the `mp4` sample: `cargo run --package client mp4 --url rtsp://ip.address.goes.here/ --username admin --password test out.mp4`

This is the output I get:

```
PS C:\Users\chriwil\source\repos\retina> cargo run --package client -- mp4 --url rtsp://localhost:8554/mystream out.mp4
    Finished dev [unoptimized + debuginfo] target(s) in 0.61s
     Running `target\debug\client.exe mp4 --url rtsp://localhost:8554/mystream out.mp4`
I20220810 14:36:31.369 main client::mp4] Using h264 video stream
I20220810 14:36:31.371 main client::mp4] Using mpeg4-generic audio stream (rfc 6381 codec mp4a.40.2)
1628175071 (mod-2^32: 1628175071), npt 0.140: 670-byte video frame
I20220810 14:36:31.537 main client::mp4] new video params: VideoParameters { rfc6381_codec: "avc1.4D401F", pixel_dimensions: (1280, 720), pixel_aspect_ratio: Some((1, 1)), frame_rate: Some((2, 50)), extra_data: Length: 43 (0x2b) bytes
0000:   01 4d 40 1f  ff e1 00 1c  67 4d 40 1f  e8 80 28 02   .M@.....gM@...(.
0010:   dd 80 b5 01  01 01 40 00  00 03 00 40  00 00 0c 83   ......@....@....
0020:   c6 0c 44 80  01 00 04 68  eb ef 20                   ..D....h..  }
E20220810 14:36:31.543 main client] Fatal: Timestamp jumped -2692 (-0.061 sec) from 2705800284 to 2705797592 (mod-2^32: 2705797592), npt -0.061; policy is to allow 0..10 sec only

conn: [::1]:52007(me)->[::1]:8554@2022-08-10T14:36:31
stream: TCP, interleaved channel ids 2-3
ssrc: 57185d5e
seq: 0000f0e9
pkt: 1762@2022-08-10T14:36:31
error: process didn't exit successfully: `target\debug\client.exe mp4 --url rtsp://localhost:8554/mystream out.mp4` (exit code: 1)
```

scottlamb, 2022-08-10:
Sounds frustrating. A few questions:

*   Does this happen immediately every time you run the example?
*   What RTSP server are you using (for a camera, make/model/version)?
*   Does adding `--initial-timestamp=permissive` to the command help?
*   Are you able to / interested in getting a packet capture with e.g. [Wireshark](https://www.wireshark.org/)? If you don't want to share it publicly with your camera's IP/MAC/etc, I'm happy to look at it privately instead. I'm wondering if e.g. the RTSP `PLAY` response has a `RTP-Info` header with a `rtptime` parameter and how that compares to the RTP timestamps in the actual data stream. (I can also add some logging around this...)

christopher-wilke, 2022-08-12:
Thx for your quick feedback.

> Sounds frustrating. A few questions:
> 
> * Does this happen immediately every time you run the example?
Yes
> * What RTSP server are you using (for a camera, make/model/version)?
I am using https://github.com/aler9/rtsp-simple-server

> * Does adding `--initial-timestamp=permissive` to the command help?
I will test this parameter by next week 👍 

> * Are you able to / interested in getting a packet capture with e.g. [Wireshark](https://www.wireshark.org/)? If you don't want to share it publicly with your camera's IP/MAC/etc, I'm happy to look [...]
I will provide you some Wireshark data as soon as possible :+1:

christopher-wilke, 2022-08-17:
Here are some updates:

Does adding --initial-timestamp=permissive to the command help?
The parameter `--initial-timestamp=permissive` did not fix the issue

I will try to provide a Wireshark protocol later

scottlamb, 2022-10-17:
Any update? I'm curious to see the Wireshark capture.

Do you know if your camera uses B-frames? This is likely the same thing as #61 if so.

astromechza, 2026-03-14:
@scottlamb I got the same when pointing this client at a stream from my Unifi Protect server.

I also tried with and without --initial-timestamp=permissive and --allow-loss flags in case that made a difference but no luck.

Commit: ff771fe

```
$ cargo run --package client -- mp4  out.mp4 --url=rtsp://192.168.1.1:7447/CBcg456njby9iXXX
    Finished `dev` profile [unoptimized + debuginfo] target(s) in 0.06s
     Running `target/debug/client mp4 out.mp4 '--url=rtsp://192.168.1.1:7447/CBcg456njby9iXXX' --initial-timestamp=permissive --allow-loss`
2026-03-14T10:29:12.511331Z  INFO main retina::codec: no depacketizer for media/encoding_name audio/opus
2026-03-14T10:29:12.511634Z  INFO main client::mp4: Using h264 video stream
2026-03-14T10:29:12.516613Z  INFO main client::mp4: Using mpeg4-generic audio stream (rfc 6381 codec mp4a.40.2)
2026-03-14T10:29:12.561241Z  INFO main client::mp4: writing MP4 failed; details will be logged with `Fatal:` after RTSP session teardown
2026-03-14T10:29:12.568236Z ERROR main client: Fatal: Timestamp jumped 1768683982 (36847.583 sec) from 0 to 1768683982 (mod-2^32: 1768683982), npt 36847.583; policy is to allow 0..10 sec only

conn: 192.168.1.66:51250(me)->192.168.1.1:7447@2026-03-14T10:29:12.504055Z
stream: TCP, interleaved channel ids 2-3
ssrc: 09e31a67
seq: 53838
pkt: 1829@2026-03-14T10:29:12.560497Z
```

I used `ffprobe` to check for pict_type "B" but I didn't capture any, just got an initial sequence of I I P I P P P P P P .... with an I thrown in every now and then. I've attached a gzipped json dump of those frame headings so you can see the timestamp info too.

[buff.gz](https://github.com/user-attachments/files/25994701/buff.gz)

<details><summary>Here's the info if that's helpful</summary>
<p>
<pre><code>
2026-03-14T10:29:55.890534Z  INFO main retina::codec: no depacketizer for media/encoding_name audio/opus
stream 0:
Stream {
    media: "audio",
    control: Some(
        "rtsp://192.168.1.1:7447/CBcg456njby9iXXX/trackID=0",
    ),
    encoding_name: "mpeg4-generic",
    rtp_payload_type: 96,
    clock_rate: 48000,
    channels: Some(
        1,
    ),
    framerate: None,
    depacketizer: Ok(
        Depacketizer(
            Aac(
                Depacketizer {
                    config: AudioSpecificConfig {
                        parameters: AudioParameters {
                            rfc6381_codec: Some(
                                "mp4a.40.2",
                            ),
                            frame_length: Some(
                                1024,
                            ),
                            extra_data: Length: 2 (0x2) bytes
                            0000:   11 88                                                ..,
                        },
                        frame_length: 1024,
                        channels: ChannelConfig {
                            channels: 1,
                            ncc: 1,
                            name: "mono",
                        },
                        audio_object_type: 2,
                        sampling_frequency_index: 3,
                    },
                    state: Idle {
                        prev_loss: 0,
[... 76 lines cut ...]
                                0010:   24 ff 80 b7  01 01 01 40  00 00 fa 00  00 30 d4 27   $......@.....0.'
                                0020:   68 22 11 a8  01 00 04 68  ee 3c 80                   h".....h.<.,
                            },
                            sps_nal: b"gM@\x1f\x9ad\x02\0$\xff\x80\xb7\x01\x01\x01@\0\0\xfa\0\00\xd4'h\"\x11\xa8",
                            pps_nal: b"h\xee<\x80",
                            seen_extra_trailing_data: false,
                        },
                    ),
                    pieces: [],
                    nals: [],
                    seen_inconsistent_fu_a_nal_hdr: false,
                    frame_format: FrameFormat {
                        h26x_framing: FourByteLength,
                        parameter_set_insertion: EachKeyFrame,
                        aac_framing: Raw,
                    },
                },
            ),
        ),
    ),
    state: Uninit,
}
</code></pre>
</p>
</details>

closed 2026-10-17 by scottlamb