scottlamb/retina #80: Failed to parse RTSP with: Invalid RTSP message

tubzby, 2023-03-23:
Thank you for your kind and nice work!

I'm testing the webrtc-proxy example, it failed sometimes with my RTSP camera (sometimes works) with messages on the console:

```console
E20230323 10:39:59.646 main webrtc_proxy] error: RtspFramingError { conn_ctx: ConnectionContext { local_addr: 192.168.8.229:65029, peer_addr: 192.168.123.140:554, established_wall: WallTime(Timespec { sec: 1679539188, nsec: 67880000 }) }, msg_ctx: RtspMessageContext { pos: 668859, received_wall: WallTime(Timespec { sec: 1679539199, nsec: 646905000 }), received: Instant { tv_sec: 228430, tv_nsec: 260722583 } }, description: "Invalid RTSP message; buffered:\nLength: 2184 (0x888) bytes\n0000:   51 f3 1c e1  24 2c 69 f1  00 4a 91 45  e2 1a bc 80   Q...$,i..J.E....\n0010:   ab 28 fe 00  cf 62 eb 48  25 72 9a fd  a9 00 c0 2a   .(...b.H%r.....*\n0020:   7e 0b b6 91  8e 32 c2 12  c2 ee 8f 7d  c6 73 09 c7   ~....2.....}.s..\n0030:   b1 41 9f f3  29 f3 59 83  0d f2 eb ed  7a 4b 82 23   .A..).Y.....zK.#\n0040:   b9 fe 29 f4  90 2d e3 35  ef ac 91 19  5e bd 2e c2   ..)..-.5....^...\n0050:   33 0c a1 35  59 b8 02 a1  67 a7 ae 34  76 cd d1 63   3..5Y...g..4v..c\n0060:   f0 f0 2f 34  73 b9 78 fa  89 dd 28 cd  67 48 b4 71   ../4s.x...(.gH.q\n0070:   43 be cd af  4b 37 76 89  24 b5 83 5e  69 03 00 18   C...K7v.$..^i...\n...2056 (0x808) bytes not shown..." }
E20230323 10:39:59.647 main webrtc_proxy] upstream failure
```

I tried to use Wireshark to analyze the stream, it might link to an invalid RTSP message, the bytes contain 3 RTSP messages, and the last one is invalid:

[image]

But it works with ffplay, so I don't know if there is any difference between ffplay and webrtc-proxy.

tubzby, 2023-03-26:
Find out that if I wait some time before inputting the SDP from the web, it has a bigger chance of causing this error.

Maybe we can connect WebRTC before connecting to the RTSP stream?

scottlamb, 2026-02-20:
I'm fixing this in the example; here's the explanation of the problem and new behavior:

```rust
    // The RTSP upstream will time out if we wait too long before sending
    // `PLAY`, and likewise will be unhappy [*] if we don't read frames quickly
    // enough. So start now. We'll discard frames until the downstream is ready.
    //
    // [*] some cameras drop the TCP connection when their send buffers fill.
    // Others truncate the current packet, which causes Retina to fail shortly
    // afterward with "framing error".
```

> Maybe we can connect WebRTC before connecting to the RTSP stream?

We do need to connect to the camera to decide what tracks we want from WebRTC. And we can't hang out too long before sending `PLAY` either as we'll time out as mentioned in the comment quoted above.

closed 2026-02-20 by scottlamb