scottlamb/retina #84: Need help to run the code

thalesmaoa, 2023-06-26:
Hello, just arrived by a suggestion of @thatdevsherry, but I'm a completely  dumb in rust. Never used and I'm having difficulties to understand its work flow. 
After some reading, I tried to run the example, but I get an error:

```
$ cargo run --package client mp4 --url rtsp://192.168.11.125:8800/ 
error: Found argument '--url' which wasn't expected, or isn't valid in this context

	If you tried to supply `--url` as a value rather than a flag, use `-- --url`

USAGE:
    cargo run --package [<SPEC>...]

For more information try --help
``` 

I have a cheap chinese brand that open using V380. I need to open it in the computer. I can find the reason for that.

thatdevsherry, 2023-06-26:
Hey, can you also try running [my forked repo](https://github.com/thatdevsherry/retina) of retina? I had tried messing around with it to support my v380 camera. If it doesn't work on upstream retina but works on my fork, then both of our cameras have the same problem.

I'm able to use my forked version to get an RTSP stream, using the following command

`cargo run --package client mp4 --url rtsp://192.168.10.10 out.mp4`

[Relevant V380 PR](https://github.com/scottlamb/retina/pull/71)

scottlamb, 2023-06-26:
> $ cargo run --package client mp4 --url rtsp://192.168.11.125:8800/ 

Sorry, typo in the README. Instead of `mp4 --url`, it should read `mp4 -- --url`. The extra `--` ends the arguments to `cargo` itself (Rust's build tool) and starts the arguments that should be passed through to the program being run.

thalesmaoa, 2023-06-29:
Sorry, but I've being trying to understand my problem, but I think this is a really simple and noob problem.

```
$ cargo run --package client mp4 -- --url rtsp://10.254.1.223:8800 --username admin --password password out.mp4
error: package(s) `mp4` not found in workspace `/home/thales/chinese_camera/retina`
```

Really need help.

thalesmaoa, 2023-07-13:
I will try as you suggested and let you know. Thanks