scottlamb/retina #102: Reolink Bad SPS

Curid, 2024-04-23:
```
Bad SPS: RbspReaderError(RemainingData)
  conn: 172.17.0.2:52680(me)->192.168.1.212:46665@2024-03-03T16:49:12
  stream: TCP, interleaved channel ids 0-1
  ssrc: 40b2a2e8
  seq: 000014b6
  pkt: 49598@2024-03-03T16:49:12
```

[Failing test:](https://github.com/scottlamb/retina/blob/main/src/codec/h264.rs#L1453)
``` rust
#[test]
fn reolink_sps() {
    init_logging();
    super::InternalParameters::parse_format_specific_params(
        "fmtp:96 packetization-mode=1;\
         profile-level-id=640033;\
         sprop-parameter-sets=Z2QAM6wVFKCgPZA=,aO48sA==",
     )
    .unwrap_err();
}
```

[reolink_working_vlc_pcap.zip](https://github.com/scottlamb/retina/files/15079974/reolink_working_vlc_pcap.zip)

danielgoz, 2024-04-23:
the camera is a Reolink Video Doorbell WiFi

scottlamb, 2024-04-24:
This camera sends an "out-of-band" sequence parameter set (in the DESCRIBE response's SDP): `Z2QAM6wVFKCgPZA=` in base64, `67640033ac1514a0a03d90` in hex. `h264_reader` says this is valid. This SPS has no VUI (which is fine).

Then the camera sends another SPS "in-band" (as part of the RTP stream): `67640033ac1514a0a03da1000004f6000063380404`. It's invalid according to `h264_reader`. `RbspReaderError(RemainingData)` means there was unexpected extra stuff after `h264_reader` parsed everything it expected to find. If I remove the last byte, it does parse. The difference between the two is as follows:

```diff
--- out-of-band sps
+++ in-band sps without last byte
@@ -1,38 +1,57 @@
 SeqParameterSet {
     profile_idc: ProfileIdc(
         100,
     ),
     constraint_flags: ConstraintFlags {
         flag0: false,
         flag1: false,
         flag2: false,
         flag3: false,
         flag4: false,
         flag5: false,
         reserved_zero_two_bits: 0,
     },
     level_idc: 51,
     seq_parameter_set_id: SeqParamSetId(
         0,
     ),
     chroma_info: ChromaInfo {
         chroma_format: YUV420,
         separate_colour_plane_flag: false,
         bit_depth_luma_minus8: 0,
         bit_depth_chroma_minus8: 0,
         qpprime_y_zero_transform_bypass_flag: false,
         scaling_matrix: SeqScalingMatrix,
     },
     log2_max_frame_num_minus4: 9,
     pic_order_cnt: TypeZero {
         log2_max_pic_order_cnt_lsb_minus4: 9,
     },
     max_num_ref_frames: 1,
     gaps_in_frame_num_value_allowed_flag: true,
     pic_width_in_mbs_minus1: 39,
     pic_height_in_map_units_minus1: 29,
     frame_mbs_flags: Frames,
     direct_8x8_inference_flag: true,
     frame_cropping: None,
-    vui_parameters: None,
+    vui_parameters: Some(
+        VuiParameters {
+            aspect_ratio_info: None,
+            overscan_appropriate: Unspecified,
+            video_signal_type: None,
+            chroma_loc_info: None,
+            timing_info: Some(
+                TimingInfo {
+                    num_units_in_tick: 1270,
+                    time_scale: 25400,
+                    fixed_frame_rate_flag: false,
+                },
+            ),
+            nal_hrd_parameters: None,
+            vcl_hrd_parameters: None,
+            low_delay_hrd_flag: None,
+            pic_struct_present_flag: false,
+            bitstream_restrictions: None,
+        },
+    ),
 }
```

so essentially the new SPS additionally declares a fixed frame rate of 10 fps (a tick every 1,270 / 25,400 seconds, 2 ticks per frame because H.264 supports interlaced video). That sounds perfectly plausible and, except for the extra byte, valid.

[... 39 lines cut ...]

    return 0;
}

// ...

static int FUNC(sps)(CodedBitstreamContext *ctx, RWContext *rw,
                     H264RawSPS *current)
{
    int err, i;

    // ...

    flag(vui_parameters_present_flag);
    if (current->vui_parameters_present_flag)
        CHECK(FUNC(vui_parameters)(ctx, rw, &current->vui, current));
    else
        CHECK(FUNC(vui_parameters_default)(ctx, rw, &current->vui, current));

    CHECK(FUNC(rbsp_trailing_bits)(ctx, rw));

    return 0;
}

```

scottlamb, 2024-04-24:
Another point of comparison: deepch/vdk just stops when it has all the data it knows to look for, [here](https://github.com/deepch/vdk/blob/4b060bc442cd5a1c6bf22f910536d5f8cae6d1ec/codec/h264parser/parser.go#L650). This program...

```go
package main

import (
	"encoding/hex"
	"fmt"
	"github.com/deepch/vdk/codec/h264parser"
	"log"
)

func main() {
	raw, err := hex.DecodeString("67640033ac1514a0a03da1000004f6000063380404")
	if err != nil {
		log.Fatalf("hex string wouldn't parse: %v", err)
	}
	sps, err := h264parser.ParseSPS(raw)
	if err != nil {
		log.Fatalf("sps wouldn't parse: %v", err)
	}
	fmt.Printf("sps: %+v\n", sps)
}
```

prints: `sps: {Id:0 ProfileIdc:100 LevelIdc:51 ConstraintSetFlag:0 MbWidth:40 MbHeight:30 CropLeft:0 CropRight:0 CropTop:0 CropBottom:0 Width:640 Height:480 FPS:10}`

Curid, 2024-04-24:
>  I think the camera is actually just sending a strange extra byte at the end for some infuriating reason.

Is there some reference H264 implementation we could try it with? Why just `0x04`, does it repeat the last byte? It'd be interesting to dump the firmware and find out. 

> Next question: what should we do about it? I'm unsure:

>    Accept that this camera doesn't work with Retina? Then we wouldn't be upholding a promise in the README: "Works around brokenness in cheap closed-source cameras."

This may not be the only camera with this issue though, and it's harder to find replacements for door bell cameras compared to normal ones.

> Make h264-parser not error if there's extra trailing data? This risks masking bugs. These bitstreams aren't exactly JSON or protobuf. They don't include the names or tag numbers for each field [...]

We could add a check for the in-band SPSs to see if the extra byte has length 1 and equals `0x04` to reduce the risk of masking bugs.

> When Retina receives an invalid SPS after having previously received a valid one, keep using the previously valid one? This is a bit similar to option 2. It could also mask problems that later [...]

If it randomly receives an invalid SPS several hours after the stream started I'd prefer if it threw an error.

scottlamb, 2024-04-27:
> Is there some reference H264 implementation we could try it with?

Turns out yes. https://www.itu.int/rec/T-REC-H.264.2-201602-I/en has a zip file with a bunch of stuff in it. Some of it seems to be specifically tailored for stuff like Scalable Video Codec. The basic thing seems to be the `H.264.2_JM_19_0` directory; the readme says the latest version can be found at <http://iphome.hhi.de/suehring/tml>. The code is...garbage quality. But you can run it as follows.

```
$ python3 -c 'open("sps-and-pps", "wb").write(bytes.fromhex("0000000167640033ac1514a0a03da1000004f60000633804040000000168ee3cb0"))'
$ CFLAGS=-g make -j
$ lldb bin/ldecod.exe -- -p InputFile=sps-and-pps
(lldb) run
```

It will crash trying to interpret video frames that aren't in the file, but it seems to parse the bogus SPS first. It doesn't care about the trailing RBSP bits being correct; it just stops reading a given NAL when it finds what it wants.

I guess that's three implementations so far that just stop reading. Should we do the same? 🤷 

> Why just 0x04, does it repeat the last byte? It'd be interesting to dump the firmware and find out.

🤷 That level of reverse engineering is more than I can do in the time I have.

> If it randomly receives an invalid SPS several hours after the stream started I'd prefer if it threw an error.

It certainly would be possible to define a policy where it only allows the invalid SPS say before the first frame of video.

Curid, 2024-04-27:
> I guess that's three implementations so far that just stop reading. Should we do the same? 🤷

You're right that it may mask other bugs, so it might be better to add a special exception for just this camera.

@dholroyd what do you think?

dholroyd, 2024-05-01:
I can see that there is an example of other syntax in the spec where it's expected to have additional data between `seq_parameter_set_data()` and `rbsp_trailing_bits()`

```
subset_seq_parameter_set_rbsp( ) {
  seq_parameter_set_data( )
  ...load of stuff...
  rbsp_trailing_bits( )
}
```

But I can't see this being needed in a normal `seq_parameter_set_rbsp()`, and I can't see any flags that would allow `seq_parameter_set_data()` to be extended in a later version of the spec (and there's a separate NAL Unit type specifically for SPS extensions).

So I agree that the example you describe seems wrong.

How would it be if `h264_reader::nal::sps::SpsError` was altered to contain those parts of the SPS that were apparently parsed prior to a syntax error?  Something like that would let the calling code make its own choices about how much to trust the data, and could also provide useful diagnostic context when trying to work out what's wrong with a bitstream.

This design still falls short on the diagnostic front because the data structures can't represent having parsed 'half' of a `SeqParameterSet` - the best that could be done is to e.g. return `SeqParameterSet` with `vui_parameters: None`  if we hit an issue in `vui_parameters( )` syntax.

Anyway, in this case, the error would contain the complete `SeqParameterSet` of course.

scottlamb, 2024-05-03:
> This design still falls short on the diagnostic front because the data structures can't represent having parsed 'half' of a `SeqParameterSet` - the best that could be done is to e.g. return `SeqParameterSet` with `vui_parameters: None` if we hit an issue in `vui_parameters( )` syntax.

How would this work if the error is before some mandatory element?

I'm leaning toward a more modest thing of just adding a Retina flag to ignore the trailing data part only. I want Retina to work with a variety of cameras, but as a general practice I prefer to understand each distinct issue to the extent I can, rather than doing something broader that might paper over some unrelated bug.

scottlamb, 2024-05-03:
I got another report in which the camera sent the SPS `67640033ac1514a02800f190` both in the `DESCRIBE` and in-band. Decoding produces `RbspReaderError(RemainingData)` unless I chop off the last two bytes. That lets us answer a couple questions:

> When Retina receives an invalid SPS after having previously received a valid one, keep using the previously valid one?

Wouldn't work; in this case it's never received a valid SPS.

> Why just 0x04, does it repeat the last byte?

In this case it's two bytes and they don't match the previous two, so apparently not.

danielgoz, 2024-05-21:
Any updates?
Anything else I could help with?

Curid, 2024-05-21:
Another report for the same error with a reolink cx410 

```
Bad SPS: RbspReaderError(RemainingData)
  conn: 172.22.0.2:57512(me)->192.168.254.104:554@2024-05-21T14:58:16
  stream: TCP, interleaved channel ids 0-1
  ssrc: dd247e4e seq: 13956
  pkt: 294297@2024-05-21T14:58:18
```

scottlamb, 2024-05-24:
> Any updates? Anything else I could help with?

This should be fixed on Retina's `tolerate-bad-sps` branch. Give it a spin? If it works, I'll prep Retina and Moonfire releases.

Curid, 2024-05-24:
cx410 and the doorbell camera confirmed working!

commit 84dcd282 (2024-05-24): Tolerate extra trailing data in SPS

I could make a knob for this, but it seems like tolerating this is
common among RTSP clients, and just doing it by default saves some
plumbing to the codec within Retina and (for Moonfire) creating config
for it.

Fixes #102
closed 2024-05-26 by scottlamb