scottlamb/retina #104: Reolink Bad SPS (new variant)

fr4nz909, 2024-05-29:
Hi Scott and folks,
sorry for the late reply, it is Franz from the E-Mails.

I was in contact with reolink support a lot while you were doing the SPS fix.
They sent me some (without any recognizable logic :D) firmware versions.

Do be clear about my setup:
- 2 reolink rlc510a bought at different days
- cam1: firmware version v3.1.0.801_22010700 -- working quite well with moonfire nvr (no SPS Errs)
- cam2: initially had version v3.1.0.2368_23062700 -- which produced SPS Errs

So Reolink Support for example sent me v3.1.0.801_22010700 - the same as is on cam1 - which unfortunately just did not work on cam2 (build and hardware NO is the same on both cams)

So now I was really happy about the fix that was pushed, but unfortunately - I just tested - there is another error (with firmware version v3.1.0.2368_23062700 - I went back to original for cam2):

`2024-05-29T20:52:21.596239  WARN tokio-runtime-worker streamer{stream="gss2-sub"}: retina::codec::h264: Ignoring trailing data in SPS 67640033ac1514a0a03da1000004f6000094d40404; will not log about trailing data again for this stream.    
2024-05-29T20:52:21.596345  WARN           s-gss2-sub streamer{stream="gss2-sub"}: moonfire_base::clock: opening rtsp://mapchat.me:555/h264Preview_01_sub took PT1.013040290S!
2024-05-29T20:52:21.596398  WARN           s-gss2-sub streamer{stream="gss2-sub"}: moonfire_nvr::streamer: sleeping for 1 s after error err=UNKNOWN: **can't load SPS+PPS: Sps(RbspReaderError(RemainingData))**`

Maybe that error tells you folks something. :)
And maybe, I could also do that if needed, the tolerance scale of retina could be increased.

Many Greets,
Franz from Leipzig

scottlamb, 2024-05-30:
Ahh, yeah, that error is enough; that code path isn't actually going through Retina but has a similar problem. I'll prepare a Moonfire fix when I can.

scottlamb, 2024-05-31:
This should be fixed with [Moonfire v0.7.16](https://github.com/scottlamb/moonfire-nvr/releases/tag/v0.7.16).

closed 2024-05-31 by scottlamb

fr4nz909, 2024-05-31:
Hey again! really cool, thanks for that. :)

When I wget the archived source code of v0.7.16, cargo test fails. Do I need some new (linux) deps for that to work?

Also, just interested, how/where did you fix this? As the moonfire code itself was not changed for the release v0.7.16?

scottlamb, 2024-05-31:
> When I wget the archived source code of v0.7.16, cargo test fails. Do I need some new (linux) deps for that to work?

It should have pretty minimal deps, and they haven't changed recently. Can you open a discussion on the moonfire repo with the details? Tests pass on CI, as you can see [here](https://github.com/scottlamb/moonfire-nvr/actions/runs/9311486111).

> Also, just interested, how/where did you fix this? As the moonfire code itself was not changed for the release v0.7.16?

It was changed, but you probably didn't see that because I'd missed pushing the changes to the `master` branch. The commit (which was on the `tolerate-bad-sps` branch and tagged as `v0.7.16`) is https://github.com/scottlamb/moonfire-nvr/commit/e6c7b800feceaed2ae124341024af8d62015867b

fr4nz909, 2024-06-02:
Works like a Charme with Reolink RLC510A now! :)))