scottlamb/retina #117: Unexpected error about paring RTP-Info param when parsing `PLAY` response.

ZXY595, 2025-08-22:
I am writing a rtsp client using retina, the rtsp stream I want to play can be played correctly by the ffplay cmd.
But I got this error: 
```
Retina error: Ok response to PLAY CSeq=4: RTP-Info param has no =\n\nconn: 192.168.1.100:48652(me)->
192.168.1.19:554@2025-08-22T15:04:30.060155727Z\nmsg: 768@2025-08-22T15:04:30.067052744Z"
```

After reading some the code, let's focus: 
https://github.com/scottlamb/retina/blob/d1e505570e00ad2a3acf4e88499ce6e1cd20a92e/src/client/parse.rs#L626C2-L630C20
And  I print `s`, it is:
`s = "url=rtsp://192.168.1.19:554/0/main/trackID=0;seq=0;rtptime=0;"`
s.split(';')

https://github.com/scottlamb/retina/blob/d1e505570e00ad2a3acf4e88499ce6e1cd20a92e/src/client/parse.rs#L672C8-L675C72
I print the `part`:
```
part = "seq=0"
part = "rtptime=0"
part = ""
```

And the `part = ""` raise the error:  `RTP-Info param has no =`.
Does this implementations is right?

scottlamb, 2025-10-26:
According to the grammar in [RFC 2326 section 12.33](https://datatracker.ietf.org/doc/html/rfc2326#section-12.33), there should not be a trailing `;`:

```
   RTP-Info        = "RTP-Info" ":" 1#stream-url 1*parameter
   stream-url      = "url" "=" url
   parameter       = ";" "seq" "=" 1*DIGIT
                   | ";" "rtptime" "=" 1*DIGIT
```

...but altering Retina to allow this anyway seems harmless enough.

scottlamb, 2025-10-26:
For the changelog, what camera/rtsp server is this?

closed 2026-02-18 by scottlamb